tenlog/TL-D3 #1: Print over USB fail

BAUBLITZ, 2019-12-27:
I have an Tenlog TL-D3S

When i print over USB then the print fails, In the log comes many resends from printed lines.
The print fails because the X-axis drive to home position and the active extruder eject the Filament.
Then the print continues without filament. This happens in any print over USB.
The same G-Code printed from SD-Card works fine .

tenlog, 2020-01-03:
We tested USB print from Ultimaker Cura 4.4, Cura 15.02 and  Repetier Host 1.0.6 on the latest formware version.

BAUBLITZ, 2020-01-11:
the print over usb seems to work. In one print the active extruder moved to its home position within the print but continued to print correctly.
Only in the logs are there still a lot of error messages with resends. It's just a beauty flaw because the print seems to continue to run smoothly, but it leaves an uneasy feeling when printing.

BAUBLITZ, 2020-01-11:
I have found this code in configuration_adv.h an i have activate it and configure it and adjusted the values as I thought, but  the print fails. What would be the right values for you that should be entered here? I search for configurations that improve the print quality

// Enable the option to stop SD printing when hitting and endstops, needs to be enabled from the LCD menu when this option is enabled.
//#define ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED

// extruder advance constant (s2/mm3)
//
// advance (steps) = STEPS_PER_CUBIC_MM_E * EXTUDER_ADVANCE_K * cubic mm per second ^ 2
//
// hooke's law says:		force = k * distance
// bernoulli's priniciple says:	v ^ 2 / 2 + g . h + pressure / density = constant
// so: v ^ 2 is proportional to number of steps we advance the extruder
#define ADVANCE

#ifdef ADVANCE
#define EXTRUDER_ADVANCE_K .0

#define D_FILAMENT 1.75
#define STEPS_MM_E 92,6
#define EXTRUTION_AREA (0.25 * D_FILAMENT * D_FILAMENT * 3.14159)
#define STEPS_PER_CUBIC_MM_E (axis_steps_per_unit[E_AXIS]/ EXTRUTION_AREA)

#endif // ADVANCE

BAUBLITZ, 2020-01-20:
Hallo with the Firmware from 20200114 the lastest the error comes back.
Not resolved

levelfifty, 2020-01-30:
I can confirm issues with USB printing, specifically using Octoprint.   Lots of Gcode resends get requested eventually it requests a line that is 50 lines behind the terminal and fails with no available history.   It appears that since the loop calls get_command_1() for the serial stream from the controller before get_command() which would be the USB, that it’s getting bogged down. The constant talking to the onboard controller doesn’t help either.  Is it possible to add some logic that puts the touchscreen controller into a suspended state or able to activate a ‘USB Mode’ so the Mega2560 can more easily handle the USB serial stream?   Commenting out #define TENLOG_CONTROLLER in configuration_zyf.h effectively fixes this issue by ignoring all touchscreen code, leaving USB as the only source of control but at the cost of no local control.

levelfifty, 2020-02-01:
Found the issue, get_command() was sending extra MSG_OK back to terminal and causing issues with the command buffer.  Submitting a PR.

Mushmuch, 2020-02-10:
@levelfifty Have you tried to print from PC or Octoprint with this change?

levelfifty, 2020-02-10:
> @levelfifty Have you tried to print from PC or Octoprint with this change?
> There is another instance of SERIAL_PROTOCOLLNPGM(MSG_OK); line 1024.

That occurance appears to only apply to commands coming from the LCD controller. (Get_command1())Since it doesn’t send full on Gcode files and only initiates SD printing, it doesn’t cause issues.   Not sure it’s entirely needed though.  

I’ve printed more than 50 hours through Octoprint since removing line 1203 under the Get_command() function, since this function is responsible for external USB printing.   With previous firmware, it would request a resend due to checksum errors every 10th or so line and eventually fail before getting 1000lines in.  After making the change, I see basically 0 resends.   That line doesn't exist in Vanilla Marlin 1.0.X, so I assume it was bug fixed out long ago from the man branch.  It was causing G1 commands to send duplicate OKs, and you know how often G1 commands happen...

Mushmuch, 2020-02-11:
I made the edit, burned the bootloader and uploaded the firmware. It's been printing from Octoprint for a couple of hours without issue now. Thanks @levelfifty! I hope that @tenlog will use Marlin 1.1.9 one day ...

chrigu70, 2020-02-12:
May i ask @levelfifty what Arduino version you use to compile the source code? Or a complete different Software?

levelfifty, 2020-02-12:
@chrigu70 I used Arduino IDE to compile and XLoader to burn the non-bootloader version.   Arduino IDE was set to Mega2560 as board type.

chrigu70, 2020-02-13:
@levelfifty  Thank you so much! I was now able to raise max_bedtemp and so far my printer (Tenlog TL-D3 pro) works fine with my modified Heatbed (220V 750W max 170°C with a 6mm Glass).
I used Arduino IDE 1.8.8 and all looks good so far...
Thank you!

tenlog, 2020-03-26:
@levelfifty Thank you, i will take your advice and fix this bug in the next version.  ;-)

Totoleheros, 2020-04-04:
@levelfifty How do you know which configuration file corresponds to your model, their names are not really informative. I have the Tenlog TL-D3 Pro (https://www.tenlog3d.com/tenlog-tl-d3-pro-independent-dual-extruder-dmp-3d-printer-with-touch-screen-support-pva-tpu-abs-pla_p11.html), which configuration file is appropriate for it? Many thanks for your help.

Totoleheros, 2020-04-04:
I have installed this firmware on my Tenlog TL-D3 Pro but it did not resolve my connection issues with my Raspberry. I have this message when I try to connect Octoprint to the printer:

There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor

Any clue?

Dane809, 2020-10-22:
My printer shut down while printing. The message on the touch panel is: "Shutting down" 
Why? Same Raspberry works perfect with my Anycubic i3 Mega

matthew-tracy, 2021-03-19:
@levelfifty @tenlog I'm still having issues printing over USB (specifically OctoPrint). I tried installing the newest firmware, and I tried commenting out every SERIAL_PROTOCOLLNPGM(MSG_OK) line, just to be safe, but I'm still getting communication errors and timeouts due to resending lines. 
Any fix?

WebSpider, 2021-09-21:
This seems to be fixed here: https://github.com/tenlog/TL-D3/blame/4300e3fb1788c24551f2ffe47ffac0a36c946852/Marlin/Marlin_main.cpp