tenlog/TL-D3 #4: Duplication mode not working from PC host

afkool, 2020-04-25:
Printer: Hands 2
Firmware: Was emailed to me
MD5SUM - 06885062a1083d88910758e12c08c2ab

How to reproduce:
Home machine
Run GCODE from PC that has M605 S2 command

P2/E2 extruder moves too far right and jams/makes.

Work around:
M605 S2
G28
Run GCODE

M605 must be done before homing.

BAUBLITZ, 2020-04-27:
Tenlog's documentation expressly states that the commands for changing the operating mode must first be in G-Gode.
M605 S1 for dual material, M605 S2 for duplication mode and M605 S3 for mirror mode

afkool, 2020-04-27:
The documentation on the sdcard or online show a screenshot of the M605 dual nozzle mode commands first yes.  However, it doesn't explicitly say that you need to home after issuing a nozzle mode change nor do the example gcode even issue a nozzle change.  

The machine shouldn't have an extruder go past the soft or hard endstops just due to switching modes.  If they don't want to fix that then that's fine but the doc should be more clear.  

I am obviously able to get past this and make great prints but it serves as good info for someone else who might get stuck.