tenlog/TL-D3 #6: Display not works with Marlin 2.0.x

BAUBLITZ, 2020-07-21:
I have update the Firmware to Marlin 2.0.x Bugfix 
but the Display do not work.
Can you share the .hmi file.

scheerjacob, 2020-08-05:
Hi @BAUBLITZ, I am trying to do Tenlog 3D pro with Klipper on OctoPrint. I have seen that you have set up your Tenlog to work with Klipper and Octoprint. Can you send me your config file? That would be great. 

Thank you so much,
Jacob

BAUBLITZ, 2021-02-01:
take a look here https://github.com/Skorpi08/Nextion4Marlin

dsz-dsz-k, 2021-02-26:
Sorry if this is an obvious question, but where does the alternate display firmware come from ? 
How can we know if it will work with our screen ?

dfgas, 2021-03-08:
Looking for firmware for the TL-D3 with 2209 drivers and full color screen the dwin model.

kva19, 2021-03-21:
Any tutorials on upgrading to Marlin 2.x? I'm am running 2209 drivers and a full color screen down model. 

Here is a link to what I purchased from Amazon.

https://www.amazon.com/dp/B08ND5ZDQW

Any help or guidance would be appreciated!

ikjl, 2021-05-16:
Is there a walkthrough for getting up to the point of display issues? 

When I tried to compile Marlin 2.0 with the configuration files for 1.0.2-2, I received "Unknown MOTHERBOARD value in configuration.h".

ikjl, 2021-05-16:
> take a look here https://github.com/Skorpi08/Nextion4Marlin

It looks like the display is for the HicTop version. Do you know what display that uses? I would swap out the display to get Marlin 2.x working. 

My TL-D3 uses a DWIN display.

tonnic, 2021-05-22:
that page being referenced doesn't say much. anyone and to give more details?

tonnic, 2021-05-23:
> > take a look here https://github.com/Skorpi08/Nextion4Marlin
> 
> It looks like the display is for the HicTop version. Do you know what display that uses? I would swap out the display to get Marlin 2.x working.
> 
> My TL-D3 uses a DWIN display.

Ya -- any luck getting Marlin 2.0.x working?   I'm almost at the point of returning this unit.  My mind is boggled that they use such an outdated version of the firmware..