tenlog/TL-D3 #7: Klipper - Thermosensor Type

scheerjacob, 2020-08-05:
Hello,

I am writing code for a configuration file to run Tenlog D3 Pro with Klipper firmware. I would like to know what the sensor type is for the thermosensor in the extruders and in the bed. Also, please let me know what pins these are connected to for Extruder 1, 2 and the bed. 

Thank you so much.

P.S. If anyone reading this has successfully run the Tenlog D3 Pro on Klipper firmware, please send us a message and your config files.

BAUBLITZ, 2020-08-06:
This is a klipperconfig for an TL-D3S it's only little different from D3 Pro. the difference can you read out from Marlin Firmware.
Thermistors types are  the same.
[printer.zip](https://github.com/tenlog/TL-D3/files/5033737/printer.zip)

scheerjacob, 2020-08-06:
Thanks, did the config file work for you in the end? I am wondering because we tried to connect our Tenlog to Klipper and Octoprint. The error we are getting is that the Octoprint is not reading the temperatures from the thermosensors on the extruders or the heating bed

BAUBLITZ, 2020-08-07:
Yes it works for me , but im back to Marlin. I miss the mirror and duplex mode in klipper. The prinquality ans speed is better in klipper and for Dualcolor is klipper very good.

vegeta340, 2021-02-01:
@BAUBLITZ please correct me if I'm wrong but "step_distance: .00974" from extruder definition is instead of esteps from Marlin? How to determine this number for Titan extruders where esteps=382.165?

BAUBLITZ, 2021-02-01:
klipper need mm/step and marlin steps/mm  382.165 in marlin are 0,00261 in klipper

matti125, 2021-05-31:
Trying to run klipper with Tenlog Hands 2, but the temperatures get reported incorrectly for the extruders:
"
Send: M105
Recv: ok B:24.0 /0.0 T0:-99.6 /0.0 T1:-99.6 /0.0
"
(To get klipper to start, I had to set the lower temp limit to -100.)

Any ideas on why this happens? Based on Tenlog https://github.com/tenlog/TL-D3/blob/master/Marlin/Configuration_xy.h, all thermistors should be of the same type, but they only the bed seems to work as expected.

tonywright, 2021-07-24:
Hi Scheerjacob, did you get klipper working with your tenlog d3 pro? I have the same issue using the above printer.cfg, it fails with ADC out of range. The pins seem to checkout ok compared to marlin firmware but I cannot see what value is being returned to see if it is anywhere close to a valid value.
Thanks

matti125, 2021-07-26:
While I do not have the Tenlog d3 pro, I have the Tenlog Hands 2, and It works fine with klipper.

tonywright, 2021-07-26:
> While I do not have the Tenlog d3 pro, I have the Tenlog Hands 2, and It works fine with klipper.

Thanks matti125, I managed to get past the temp warning by setting the minimum to -100 but after that the printer wouldn't move any of the steppers although it seemed to be connected ok. Might give it another go in a bit and recompile the bin file just in case I mixed that up.

marcello-fialho, 2021-08-15:
> While I do not have the Tenlog d3 pro, I have the Tenlog Hands 2, and It works fine with klipper.

Do you have the printer.cfg for the Hands 2? Is it possible to do the dual z synch with klipper?

matti125, 2021-08-18:
AFAIUI, klipper does dual z synch when you you have two z axis definitions. Here are my config files. The second head is currently not xy calibrated, so ignore the x and y offsets.

printer.cfg:
```
[include kiauh_macros.cfg]
[include macros.cfg]
[printer]
max_velocity: 200
max_accel: 4000
#max_accel: 7000
#max_accel_to_decel: 7000

#[include extruders_titan/printer.cfg]
#[include extruders_std/printer.cfg]
[include extruders_titan/extruder0_titan_maxiwatt]
[include extruders_titan/extruder1_titan_stdheatblock]

[mcu]
baud: 250000
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
pin_map: arduino

[resonance_inducer]
probe_points: 110,110,30

[pause_resume]

[display_status]

[virtual_sdcard]
path: ~/gcode_files

[printer]
kinematics: cartesian
max_z_velocity: 20
max_z_accel: 60

[input_shaper]
shaper_freq_y: 35  # frequency for the Y mark of the test model

[bed_screws]
screw1: 25,35
screw2: 25,185
screw3: 195,185
screw4:	195,35

[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
rotation_distance: 40
microsteps: 16
endstop_pin: ^!ar3
position_endstop: -48
position_max: 220
position_min: -50
homing_speed: 40
homing_positive_dir: false

# Definition for the secondary carriage and extruder1
[dual_carriage]
axis: x
step_pin: ar36
dir_pin: ar34
enable_pin: !ar30
rotation_distance: 40
microsteps: 16
endstop_pin: ^!ar2
position_endstop: 265
position_min: 0
[... 251 lines cut ...]
        {% set z_safe = max_z - act_z %}
    {% endif %}
    ##### end of definitions #####
    SAVE_GCODE_STATE NAME=PAUSE_state
    BASE_PAUSE
    G91
    G1 E-{e} F2100
    G1 Z{z_safe}
    G90
    G1 X{x} Y{y} F6000

[gcode_macro RESUME]
rename_existing: BASE_RESUME
gcode:
    ##### set defaults #####
    {% set e = params.E|default(1) %} #edit to your retract length
    G91
    G1 E{e} F2100
    G90
    RESTORE_GCODE_STATE NAME=PAUSE_state MOVE=1
    BASE_RESUME

################################################################################
################################################################################
```

matti125, 2021-08-18:
And in case you run with standard hotends/extruders, here is the cfg file for those. Most importantly it has a different extruder feed ratio.
```
[printer]
max_velocity: 200
max_accel: 7000
#max_accel: 7000
max_accel_to_decel: 7000

# The definition for the primary extruder
[extruder]
step_pin: ar57
dir_pin: !ar58
enable_pin: !ar59
microsteps: 16
rotation_distance: 32.5
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar11
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog15
pressure_advance = 0.1
min_temp: 0
max_temp: 260
max_extrude_only_distance: 200.0

[extruder1]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
microsteps: 16
rotation_distance: 32.5
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
pressure_advance = 0.1
min_temp: 0	
max_temp: 260
max_extrude_only_distance: 200.0

[input_shaper]
shaper_freq_x: 34  # frequency for the X mark of the test model
```

tonywright, 2021-08-18:
Hi Matti125, thanks for uploading your configs, I really hoped this would help But I have exactly the same problem, after lowering all 3 minimum temperatures to -200 the system appears to work right up until I try moving a servo, then it just times out without a glimmer of movement.
I can read the status: "ready", I can read and trigger the (Optical)  end stops, the LEDs on the end-stops are lit so it appears to be getting power. That is the only indication though as it has a momentary power switch without an "On" position.   all the thermostats are reading -99.6 degrees. I am just wondering if there is a board pin that needs to be "switched" on to put power to the motors?

Klipper was so easy to get working on my Ender 3 and I much prefer it to the out-dated version of Marlin on the tenlog.

Thanks

Tony

matti125, 2021-08-18:
Hmm.. are your VGA cables properly connected to the extruder/hotends? And you have checked that the pins are correct on the printer.cfg for measuring the temperature and controlling the motors? And then of course, do you have the power switched on for the printer? The control board will power on with the just power from USB (from the RPi), so it will appear working, even though no motors would then really step/spin. Sorry, basic stuff, but I was. myseld hit by no having one of the VGA connectors  properly plugged in, which resulted in the temp readings being negative.

tonywright, 2021-08-18:
I don't think it is a cable problem as I have switched back and forth with Marlin and it works fine with that. I have double and triple checked the pins and they match the pins.h (pins_TENLOG_D3_HERO.h) in the marlin firmware, the only difference is that in klipper they have for example "analog15" for the temp sensor pin instead of just "15" as in the pins.h file.
I now have the display showing the "Loading" screen when it is powered up and this goes off if I power it down using the switch so I think the main power is on.
I might try putting a modified copy of Marlin 2.x on it that I have found. Unless I start feeling very adventurous and replace the mother board with a btt skr1.2 pro :) I know Klipper will work with that.

Thanks for the help.

matti125, 2021-08-18:
I guess your pins are then exactly the same ones I have?

I might still want to check that the power supply is really outputting the 24V high power lines. It might quite well be that there is a pilot +5v voltage that is always on after toggling the on/off switch (regardless of any software commands), but the 24 would need to be switched on by the software switching on a pin. Did u check whether there is a something on the power supply terminals, labeled as "control" or similar? Something that would probably have a thin wire going to it, from one of the output pins of the control board. (Btw, I used to design power supplies way, way back  :).
At least the board I have is suitable for klipper, my standard printing speed is now 100mm/s with input shaping, pressure advance and custom python-based functionality. Using Marliln would seem like a step back :/

tonywright, 2021-08-18:
I don't know what happened the first time but just tried it again and the display is powered by usb, if I click "Home" and immediately turn on the power switch the heads move for a few seconds until the power automatically goes off
I just need to find the pin number that keeps the power on, I don't really want to start messing with the hardware until I rule everything else out.
Looking in marlin and the "#define PSU_CONTROL" is remmed out so it looks like it might be else where, unless the power board is checking something else that happens when Marlin starts and then keeps the power on.
Mind you if I can find the thin wire you mentioned I could trace it back to the motherboard and perhaps find the pin number that way. The printer is in a cabinet and it is very difficult to get the bottom panel off.

tonywright, 2021-08-18:
I have found this in the tenlog pins.h file:
#define PS_ON_PIN 40 //zyf 40		//PF1

I then found this page https://github.com/KevinOConnor/klipper/issues/1295

that gave me this code that I added to printer.cfg:
[output_pin ps_on]
pin: !ar12
value: 0
shutdown_value: 0

[gcode_macro m80]
gcode:
  SET_PIN PIN=ps_on VALUE=1

[gcode_macro m81]
gcode:
  SET_PIN PIN=ps_on VALUE=0

I changed the pin to ar40, now clicking M81 macro turns on the power and M80 turns it off.
With power, everything works including the thermostats. :)

Thanks for the help in getting klipper working with my tenlog d3 pro.

marcello-fialho, 2021-08-19:
Yeah I have the titans too. Thanks a lot! I wanted to do the upgrade to rep rap but ended up never doing it. Klipper seems a better option, considering how much the community has grown (I know it doesn't do duplication or mirror mode, but I never use those features).

matti125, 2021-08-19:
@tonywright , glad you were able to fix it! And @marcello-fialho, as u said, there is no mirror printing (yet) with klipper, but I rather find it a gimmick instead of a really useful feature. Otherwise, the concept behind klipper is just so much better IMHO. And if you really, really wanted to do mirror printing, I guess you could write you own stuff in python to do that :)

marcello-fialho, 2021-08-19:
I intend to try using the basic building blocks of digital electronics, logic gates. Since the Tenlog board drives steppers via step/dir and they are both logic level pulse train signals it should be doable. Like you said, it is a gimmick.

marcello-fialho, 2021-09-04:
As for duplication and mirror mode, tou were right, the developer with the alias Tircown already merged some pull requests related to mirror and duplication mode, it should be in the next release. They seem to do at least one release per year so it should be out before the end of 2021

dfgas, 2023-03-22:
did you guys ever get the tl-d3 pro with the 8bit board working and has the mirror and duplicate modes been added yet?

tenlog, 2023-03-22:
Our TL-D3 Pro is configured with an 8 bit board. All of our machine models
can print 4 patterns. Mirror mode, two color mode, copy mode, monochrome
mode. Is there anything else that can help you?

kiler129, 2023-03-22:
@tenlog The firmware shipped is an ancient Marlin v1.
@dfgas There's a config for Klipper but I don't think Klipper supports mirror/duplicate modes, unless you tinker with GCode in your slicer.

tenlog, 2023-03-22:
Our printer comes with 4 printing modes to choose from. Click this position
on the screen to set it up, can you find it?

dfgas, 2023-03-22:
> […](#)

we are talking klipper, not your firmware.....

tenlog, 2023-03-23:
Can you send me a picture of your printer's screen setup?

shablam, 2023-04-26:
> 

HI would you mind sharing your printer.cfg for the tl d3 pro that you got working? This one is not working for me.  https://github.com/jonpackard/klipper_config_tld3pro/blob/master/printer.cfg something must be off on the pins. thanks in advance.

kiler129, 2023-04-26:
@shablam https://github.com/Klipper3d/klipper/pull/6036/files

shablam, 2023-04-26:
> @shablam https://github.com/Klipper3d/klipper/pull/6036/files

HI there. i tried that file and am new to Klipper. but it homes ok then steppers keep grinding once z axis is down. you ahve that issue? i just used the Home button.

shablam, 2023-04-27:
> @shablam https://github.com/Klipper3d/klipper/pull/6036/files

ok thanks again. I did get it working but for my BMG unit some values had to change. Overall yours worked for me less then rotation distance mainly. also 115200 rate.

RagDollino, 2023-04-28:
> > @shablam https://github.com/Klipper3d/klipper/pull/6036/files
> 
> ok thanks again. I did get it working but for my BMG unit some values had to change. Overall yours worked for me less then rotation distance mainly. also 115200 rate.

Hello, what values ​​did you change for the BMG extruders? I'm a bit new to klipper... Thanks in advance.

shablam, 2023-04-29:
> Hello, what values ​​did you change for the BMG extruders? I'm a bit new to klipper... Thanks in advance.

HI, mainly the rotation. that must be for a stock extruder. the rotation is more like 7.9. i have it around 9 but the calculations I am told put it about 7.9. So i probably will change it. the other link I posted has some numbers I used. also change the baud rate to 115200.