tenlog/TL-D3 #9: AXE Y doesn't work !!!!!

studiolab39, 2020-10-16:
During the homing the Y axe doesn't work !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
This since a long time and the oldest firmware work so why ? Where is the problem ???????

Franck

Dane809, 2020-10-20:
Hey, same Problem

ajevans99, 2020-10-22:
I'm also experiencing this issue. I want to test this with @Dane809 's suggestion, but I don't understand how to create the `.hex` file. I get this when I try to run the make file. Anyone know what I'm missing here? Are there instructions somewhere?

```bash
➜ make                        
make: *** No rule to make target `applet/wiring.o', needed by `applet/Marlin.elf'.  Stop.
```

studiolab39, 2020-10-22:
Ok work now .... go to the configuration_Tenlog.h
and make the job ...
[image]

studiolab39, 2020-10-22:
Also you can change the extruder direction, validate or now TMC 2208, 2225, 4988  in the same tab !

DU4FDR, 2020-11-17:
Bonjour,
merci Franck
a+ Francis