tenlog/TL-D3 #11: Source Code for display is missing

iplayfast, 2020-10-22:
Source code for display is missing.

studiolab39, 2020-10-22:
i need it too .....

jpcwebb, 2020-12-01:
I emailed Tenlog about this - they will not provide the source code for the LCD as they say it is proprietary. They will also not provide schematics or pinouts for the motherboard. All in all, rather pointless them making it so-called open source when only providing partial information.

studiolab39, 2020-12-01:
Hi,

 

I am agree with you and i have the same problem with them …
i asked them the same things and they told me : we don’t have !!!

It's very regrettable …

 

Sincères salutations

Franck BULLOT

 <mailto:Studiolab39@gmail.com> Studiolab39@gmail.com

0651460428

Mikhael-Danilov, 2020-12-26:
It would be extremely useful to have display firmware available as open source.