tenlog/TL-D3 #12: lcd calibrage erroné

DU4FDR, 2020-11-16:
Hello,
I have a problem with the LCD display the calibration is wrong, impossible to use?
despite a reset and calibration the error is present.
How to empty the clibration recording

AlexLysov, 2020-12-01:
> Hello,
> I have a problem with the LCD display the calibration is wrong, impossible to use?
> despite a reset and calibration the error is present.
> How to empty the clibration recording

Do you solve the problem?

DU4FDR, 2020-12-01:
Hi,
I found the solution, a HEXA file is available to correct the LCD defect. The LCD works again.
Thank you

closed 2020-12-01 by DU4FDR

AlexLysov, 2020-12-02:
Well done!
If it possible, describe a more detailed procedure. I will be infinitely grateful! Thankyou!

AlexLysov, 2020-12-02:
Technical support offered to update the firmware, I'm afraid that their file does not reset the screen calibration

DU4FDR, 2020-12-02:
Hi
Just put it on a micro sd card and place it in the video card reader (you have to open the bottom of the printer to access).

 

then we start the printer or better we plug the printer cable into the pc the program starts and erases the firmware from the screen.

 

You remove the card and restart the printer or reconnect the cable, the calibration screen appears and you can recalibrate the touch screen at several points and several times.

 

at the end we reload the firmware of the screen via the micro sd card CAUTION only place a .tft file on the card.

 

Here is if it can help ....

 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

AlexLysov, 2020-12-02:
Thank you so much. I will try to do it today!

DU4FDR, 2020-12-02:
I have the project on the old motherboard V1 of the TLD3 to activate the automatic shutdown of the 220Volts power supply.

What do you think ?

-- 
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus