tenlog/TL-D3 #15: Extruders Freezing Up! Is this happening to you?

rad-spider, 2021-02-09:
Is an ongoing issue that will happen at random after five hours. At first I thought it was a faulty VGA cord. Or maybe I just wanted to believe it was. I know it's not my hotend. The extruder causes the clogs cause it freezes up as if there is no power going to it! So I reset the machine, and the extruder works again. And it hasn't been limited to extruder one. I've seen it happen to the second one. One extruder will work fine, then the other will freeze up mid print. So it's not always the same one.

Sent an email to Tenlog support about it and they just ignored me (like usual, unless it's to complete a sale - thanks Tenlog -_-). Trying to work out what it could be. I think it might be either software or motherboard related. 

I saw in another post a guy felt the motors weren't getting enough power. Could this potentially be a reason for the extruder motors failing?

rachael7, 2021-02-09:
Sounds like a motherboard problem to me. Maybe the heaters are cutting out and causing the clog? Either way, heater or extruder motors, it sounds like a motherboard issue, not software. I've had personal experience with 4 of them and I follow the FB group, where there are LOTS of problems, and I've not seen this issue come back to software. Unless there's something really weird with your slicer producing bad gcode, I'd say you're looking at a motherboard.

rad-spider, 2021-02-09:
I was afraid of that. Well, hopefully I can upgrade the motherboard down the line. I don't really want to buy a replacement board when the performance of the original board has been this disappointing. Especially since I've had this machine for five months or so. 

I appreciate your input. 

I'll probably run a few more tests when I have the time, but I think your right. It's the stock board.

rachael7, 2021-02-09:
Yeah, I think that's the right approach. The stock motherboard really is garbage, both in quality and the performance. The failure rate is enormous on them, and even when they work, the performance is middling at best.  If you use Facebook, there's a group on there called the MarX group that is working with Duet boards on the Tenlog machines. There is also some discussion, though not yet step-by-steps, for the BigTree Tech GTR V1.0 board.  Good luck!

jpcwebb, 2021-02-09:
can't find that group - can you post a link?

rachael7, 2021-02-09:
Sure. Here you go: https://www.facebook.com/groups/237163797566846

jpcwebb, 2021-02-09:
Thanks - just kept coming up with lots of groups about Marxism! :D

BIGMATTDAWG, 2021-07-24:
i am currently having extruder freeze issues

Pliskinware, 2021-07-29:
Is there really no hope for this issue besides upgrading? I made the mistake of purchasing this as my first personal printer (after working on professional printers for years) and keep getting hit with this issue on extruder 1 only. It's infuriating bc it makes me too uncomfortable using it over extruder 2, so I basically only have 1 extruder now.

rad-spider, 2021-07-30:
Been a long time since I talked about Tenlog. I did end up learning it was in fact the motherboard. Any minor dip or rise in power along the circuit (mine shared the same circuit as a microwave), the extruder freezes. Not just extruder one, I learned it can happen to the other as well if it's active. The only fix is to turn it off and on again. Then for me, I usually had to clean out the nozzle. Although I did eventually learn the machine stripped a gear in the extruder from all this nonsense.

The short of it is the motherboard is junk. The only workaround is you make sure the printer is on an outlet where the likelihood of the power fluctuating on that circuit is low (which isn't a guarantee it'll be fine). Otherwise you can upgrade it to a Duet or a board of your choice. Presently, I'm not interested in my Tenlog. My new passion project is a Core XY I bought second hand. I continue to tinker with it.  It's been a bit of pain, but the frame is high quality, it's easier to upgrade, and I already have the Duet boards from back when I intended to put them in my bedslingers. Still, I can't really complain about the Tenlog's frame as far as bedslingers go. It's pretty decent. But I'm probably going to sell it after I restore it to be like new. I'm not saying you need to do the same. I'm just giving you the update.  There are many who convert their bedslingers into better machines by upgrading parts.

Ultimately it's up to you on what to do. I'm just stating that the default motherboard is bad. Although, I'm wary of the extruders as well now. The gears are low quality and the boards controlling them are exposed. Should you upgrade, it maybe a good idea to discard those control boards and upgrade the extruders.

I wish you the best of luck. I know what it's like to feel like having made a mistake with my own first 3D printer. All we can take away from them is the value in the lessons they've taught us. I hope this helps you.

alikhorshidi, 2023-05-01:
Attention: I've figured out what my problem was! This is a final fix!! IT WAS BECAUSE OF SVGA CABLE/WIRING! You should wire your E1/E2 motors externally and you'll see the problem will be fixed! I've tested more than 300Hrs. print without a problem! Just attach a connector to the wire and solder it to your E1/E2 driver board on your machine. Keep SVGA cables just for extruder heating/sensor etc. CONNECT/DRIVE YOUR E1/E2 MOTORS WITH EXTERNAL WIRES!

CastleDeBirdo, 2023-06-10:
> 

Could you supply a picture to show how you did this? I'm at my wits end with this printer

amayedo, 2025-04-12:
> Attention: I've figured out what my problem was! This is a final fix!! IT WAS BECAUSE OF SVGA CABLE/WIRING! You should wire your E1/E2 motors externally and you'll see the problem will be fixed! [...]

please could you share your Fix?. I think is a great fix. But I would like to see how you did it. thanks