tenlog/TL-D3 #19: Mesh leveling

dfgas, 2021-03-07:
Please add mesh leveling. It would help with people's warped beds from factory. 9 point would be perfect.

doppelgrau, 2021-03-22:
Yes, would also really like that.
Had it wit my old i3 Mega (I think was more than 9 points), and it really helped getting good first layer.

chilternflyer, 2021-05-22:
Come on!! Why is there no Mesh Bed levelling on the TL-D3?????
dfgas is right.... These machines have big beds on them and mine is NOT flat! :-(

Hugo916, 2022-01-10:
I would also really need that. My beds are also not flat. 
So i don´t understand, why Mesh Bed levelling in not added.
Half a year has passed since the last post from chilternflyer, but still nothing has happened.
This is really a very important feature for such large machines with a punched bed.
So please build that in for your customers ;-)

captaincreative, 2022-01-20:
there is a way to auto bed level but i cant get the firmware to work - printer stuck on loading screen and never boots