tenlog/TL-D3 #27: No detection TLJ

FredericDem, 2021-11-01:
Since version V 1.0.30 my machine starts beeping continuously at startup, only solution stopped ..... I once had this problem when compiling with the DWN screen by mistake ... so I think my TLJ screen is not recognized ....

Depuis la version V 1.0.30 ma machine se met a bipper en continu au demarage seule solution l arretée ..... j ai une fois ce probleme en compilant avec l ecran DWN par erreur ... donc je pense que mon ecran TLJ n est pas reconnu ....  Une idée

FredericDem, 2021-11-01:
[image]
c est plutot ca le probleme ..

aelberone, 2022-08-14:
Probably the simplest workaround, for TJC Screen, is to swap values in _pins.h_ to:
``` ino
#define BEEPER_OFF HIGH
#define BEEPER_ON LOW
```
(TJC Beeps on `LOW' !)