tenlog/TL-D3 #30: shutting down and restarting with usb connected

kdg3737, 2022-05-05:
When I shut down this printer using the power switch and usb is still connected it says 'shutting down' on the display, but it never actually does shut down (hangs). Turning the power back on it displays some chinese characters I don't understand and it won't start or get into the main menu (hangs on the Chinese idk what). I tried sending M0 and M81 gcode commands to shut it down but for some reason it moved the z axis all the way up causing the motors to skip...

I was planning to use a dedicated PC or MCU which will always power the usb, so I need some way to shut down and restart the printer. Sending some gcodes to start-stop would be great but at this point I'd already be happy if the power switch could work as expected and not hang the printer.

How to properly shut down and restart this printer with usb connected? Is there a list of supported gcodes somewhere?

tonywright, 2022-05-05:
What you can do it alter the usb plug/cable so that the 5v is not connected, just the data lines and ground. Then when you power off with the switch it will actually go off.
At the moment the 5v from the usb will keep the board and display on.

kdg3737, 2022-05-05:
Cutting the 5V would work and I'll probably end up doing just that. But I was hoping there were other more elegant ways to recover from this. After a power failure I'd also like to be able to resume printing where it left off.

kdg3737, 2022-05-05:
I measured 4.2V on the USB port with nothing attached. I'm guessing there's a diode between that and the 5V of the printer PSU. I might be able to power a small arduino with that 4.2V and that should also fix my issue.

Do you know by any chance how many amps I can draw from that USB port @tonywright? I have some range sensors that draw ~300mA@5V.

tonywright, 2022-05-05:
Sorry, not sure the available amps from this usb but probably whatever the standard is for usb.  
How will having an Arduino in between fix your problem? Will it just have a relay/transistor to cut the 5v from the usb ?